kike-canaries/canairio_firmware #63: Mapa en InfluxDB usando el plugin Worldmap Panel.

danielbernalb, 2020-07-08:
Failed to map Grafana with the current database hosted on the InfluxDB.
In our case we have the nex format
[image]
and Grafana can't seem to read the complete table.
Anyone know how to get the Worldmap plugin to read the database formatted like this?

hpsaturn, 2020-07-08:
Please clarify the issue with the dataset, what is the reading issue, please add output here maybe?

hpsaturn, 2020-07-08:
Testing the next query, the current data structure are ok for `Worldmap plugin`:

[image]

danielbernalb, 2020-07-08:
> Testing the next query, the current data structure are ok for `Worldmap plugin`:
> 
> [image]

Thank you, great job

hpsaturn, 2020-07-09:
I found the 'alias' workaround in the next thread:

https://community.grafana.com/t/influxdb-and-grafana-plugin-worldmap-panel/16761

hpsaturn, 2021-08-25:
The last version of CanAirIO have automatic publication to world map in Grafana.

closed 2021-08-25 by hpsaturn