kike-canaries/canairio_firmware #107: TTGO TDisplay - Plantower sensor not detected

logiclabs, 2021-06-17:
I've been trying the latest firmware release v793 for the TTGO-TDisplay but with a Plantower PMS5003 sensor.  The sensor is not being detected on the RX/TX pins 12/13 (as per the pins in the sensor lib). Everything else appears to work great.

I'm guessing this is something in the build, like missing the correct UART pins for TTGO-TD?  

I also attempted to build the ttgo-tdisplay branch and set the RX/TX pins as build settings, and in this build the sensor works OK, but it appears to be pretty old and incomplete (very rough UI).   

Is it right to assume the latest TTGO-TDisplay code hasn't been pushed to github or merged to main yet?

hpsaturn, 2021-06-18:
Hi,

Thanks for writing, it's true, the TTGO T-Display code right now is in development in private a branch, Only some patrons and donors have access to it. I'm waiting to reach some goals and have a more stable version for release and merge with master. For other side the current TTGO T-Display firmware released only have enable i2c sensors detections, I could compile a version with UART detection for you,  to close this issue.

logiclabs, 2021-06-18:
That would be great.  Thank you for the quick response and your great work so far. From what I've seen, the TTGO version looks very slick.

hpsaturn, 2021-06-19:
A few minutes ago I released on dev channel the last firmware for CanAirIO Bike (T-Display flavor) with UART detections enable, if you T-Display has internet (WiFi configured via CanAirIO app), it should be have a new remote update (FOTA). With that in theory your Plantower should be works, but with UART detections enable the boot time is more slowly, a few seconds more. Please let me know if this works.

logiclabs, 2021-06-19:
Thank you!  Yes, I re-flashed r797 and the Plantower is now working fine and is detected quickly, with no start delay (only a 2-3 second delay if it doesn't find it). I did have to swap RX/TX pins, but that is probably just my misunderstanding.

Thanks for the quick turnaround.

hpsaturn, 2021-06-19:
Good! I released a new version r798, some minutes ago, because with Panasonic version we found a bug, maybe if this update works fine, you could let me know? Thanks.

closed 2021-06-23 by hpsaturn