kike-canaries/canairio_firmware #115: Flashing ESP32DEV

Geelhem, 2021-08-22:
## Steps to Reproduce

I flashed the device with the wrong board firmware  TTGO_T7 instead of ESP32DevKit  (via platform io on MAC)
I have tried to reflash on top but it does not boot! Do I need to factory reset first if so how?

## Logs
 canairio_firmware % pio run -e ESP32DEVKIT --target upload 

Processing ESP32DEVKIT (platform: espressif32; board: lolin32; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin32.html
PLATFORM: Espressif 32 (3.3.1) > WEMOS LOLIN32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 61 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.18.3
|-- <ESP8266 Influxdb> 3.8.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|-- <esp32FOTA> 0.1.4+sha.d1bbc3f
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.3
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|-- <Adafruit Unified Sensor> 1.1.4
|-- <Adafruit AM2320 sensor library> 1.1.4
|   |-- <Adafruit Unified Sensor> 1.1.4
|   |-- <Wire> 1.0.1
|-- <Adafruit BME280 Library> 2.2.0
|   |-- <Adafruit BusIO> 1.8.3
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|   |-- <Adafruit Unified Sensor> 1.1.4
|-- <Adafruit BME680 Library> 2.0.1
|   |-- <Adafruit BusIO> 1.8.3
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|   |-- <Adafruit Unified Sensor> 1.1.4
|-- <Adafruit BusIO> 1.8.3
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Adafruit SHT31 Library> 2.0.0
|   |-- <Adafruit BusIO> 1.8.3
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <AHT10> 1.1.0
|   |-- <Wire> 1.0.1
|-- <DHT_nonblocking> 1.0.0+sha.999fd0d
[... 173 lines cut ...]
Writing at 0x00162bfa... (78 %)
Writing at 0x00168d7e... (79 %)
Writing at 0x0016e74d... (81 %)
Writing at 0x00174607... (82 %)
Writing at 0x0017a188... (84 %)
Writing at 0x0017ff17... (85 %)
Writing at 0x001859b2... (87 %)
Writing at 0x0018b656... (89 %)
Writing at 0x0019172a... (90 %)
Writing at 0x001982a1... (92 %)
Writing at 0x0019e363... (93 %)
Writing at 0x001a4782... (95 %)
Writing at 0x001aad02... (96 %)
Writing at 0x001b1179... (98 %)
Writing at 0x001b6c68... (100 %)
Wrote 1746592 bytes (1041615 compressed) at 0x00010000 in 15.0 seconds (effective 929.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
========================================================================================== [SUCCESS] Took 26.98 seconds ==========================================================================================

Environment    Status    Duration
-------------  --------  ------------
ESP32DEVKIT    SUCCESS   00:00:26.981

hpsaturn, 2021-08-22:
Maybe do you have the output log? or why you mentioned that doesn't boot? 

Also try to upload it via binaries files, like this:
https://canair.io/docs/firmware_upload.html#canairio-installer

hpsaturn, 2021-08-22:
Maybe if you need clear complete settings on the ESP32 board, you can use `esp_tool` command for clear flash memory, or you can use the CanAirIO installer command like this:

```bash
./install.sh erase /dev/ttyUSBX
```

closed 2022-02-05 by hpsaturn