kike-canaries/canairio_firmware #123: [CORE PANIC] after two hours on T7+SPS30+DHT22 it reboot

hpsaturn, 2021-09-13:
After two hours on T7+SPS30+DHT22 it reboot

```
11:55:02.730 > Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
11:55:02.735 > Core 0 register dump:
11:55:02.738 > PC      : 0x40094153  PS      : 0x00060433  A0      : 0x80093627  A1      : 0x3ffbd130  
11:55:02.747 > A2      : 0x3fff7c18  A3      : 0x3ffbd2b4  A4      : 0x00000001  A5      : 0x00000001  
11:55:02.752 > A6      : 0x00060423  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffbab84  
11:55:02.760 > A10     : 0x3ffbab84  A11     : 0x00000000  A12     : 0x00060a20  A13     : 0x00000001  
11:55:02.769 > A14     : 0x00060a23  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
11:55:02.777 > EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
11:55:02.785 > 
11:55:02.785 > ELF file SHA256: 0000000000000000
11:55:02.788 > 
11:55:02.788 > Backtrace: 0x40094153:0x3ffbd130 0x40093624:0x3ffbd150 0x40091d4b:0x3ffbd170 0x40091dd9:0x3ffbd190 0x40093c96:0x3ffbd1b0 0x40093d77:0x3ffbd1f0 0x400915e2:0x3ffbd220
11:55:02.802 > 
11:55:02.802 > Rebooting...
``` 

**To Reproduce**
Steps to reproduce the behavior:
1. Config: TTGO T7 rev823 + SPS30 UART + DHT22
2. Settings: i2c force off, stime 5seg, BLE disconnected, Debug OFF, WiFi configured and working.
3. After two hours the PANIC success

roberbike, 2021-09-13:
Sensor GCJA5 i2c
i2c force: on
Debug: on
This is what the device looked like

roberbike, 2021-09-14:
After check all, The device was re-flashed again.  Now is working properly.
Conclusion: The firmware was corrup in the device.

hpsaturn, 2021-09-14:
ok, it is weird because the firmware upload flow has validation.. Anyway, I'm going to close the issue..

closed 2021-09-14 by hpsaturn