kike-canaries/canairio_firmware #125: As the sampling time increases, faults begin to occur

arttesano, 2021-09-23:
Thanks for help up and contribute to CanAirIO project. Please edit and fill this template.

## Description
When I have the sensors in small sample rates, 5 to 30 seconds there are no disconnection failures or measurements that stay in a single value,but when I go up to values of 60 or 180 seconds, disconnection faults begin to appear or measurements appear at the same value

## Debug output

If apply, please enable debugging from the CanAirIO app on `settings->mobile station->enable debug mode` and paste the output.

## Device config

```
board:  ESP32DevKit
firmware version: rev827 (prod)
mobile app version: rev5xx
sensors: SPS30 UART
OLED: no
TFT: no
Power: phone charger
```

## To Reproduce
increase the sample rate value

## Additional context

http://influxdb.canair.io:8000/d/GU68_kcMz/banco-prueba-arttesano?orgId=1&from=1632343527895&to=1632386727895&refresh=1m&viewPanel=19

arttesano, 2021-09-23:
-->[CONF] request: {
  "act": "rbt",
  "cmd": "47179",
  "aenb": false,
  "denb": false,
  "i2conly": false,
  "ienb": false,
  "wenb": false,
  "wmac": 0,
  "wsta": false
}
-->[CONF] reboot..
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

== CanAirIO Setup ==

-->[CONF] debug is enable.
-->[OGUI] display config ready.
-->[INFO] ESP32MAC: 98:F4:AB:6D:4B:BA
-->[INFO] Revision: r827
-->[INFO] Firmware: 0.4.3
-->[INFO] Flavor  : TTGO_T7
-->[INFO] Target  : prod
-->[INFO] Detecting sensors..
-->[INFO] PM sensor configured: 2
-->[SLIB] new sample time: 1
-->[SLIB] debug is enable.
-->[SLIB] temperature offset: 0.00
-->[SLIB] altitude offset: 0.00
-->[SLIB] forced only i2c sensors: 0
-->[SLIB] UART detecting SPS30 PM sensor.. 
-->[SLIB] UART SPS30 starting sensor.. 
Sending:  0x7E 0x00 0xD1 0x00 0x2E 0x7E
Received: 0x7E 0x00 0xD1 0x00 0x07 0x02 0x02 0x00 0x07 0x00 0x02 0x00 0x1A 0x7E length: 13

Sending:  0x7E 0x00 0xD0 0x01 0x03 0x2B 0x7E
Received: 0x7E 0x00 0xD0 0x00 0x11 0x39 0x36 0x30 0x44 0x31 0x30 0x46 0x32 0x36 0x37 0x46 0x35 0x46 0x43 0x42 0x36 0x00 0x79 0x7E length: 23

-->[SLIB] SPS30 Serial number :  960D10F267F5FCB6
Sending:  0x7E 0x00 0xD0 0x01 0x00 0x2E 0x7E
Received: 0x7E 0x00 0xD0 0x00 0x09 0x30 0x30 0x30 0x38 0x30 0x30 0x30 0x30 0x00 0x9E 0x7E length: 15

-->[SLIB] SPS30 product name  :  00080000
Sending:  0x7E 0x00 0xD1 0x00 0x2E 0x7E
Received: 0x7E 0x00 0xD1 0x00 0x07 0x02 0x02 0x00 0x07 0x00 0x02 0x00 0x1A 0x7E length: 13

-->[SLIB] SPS30 firmware level:  2.2
-->[SLIB] SPS30 Hardware level:  7
-->[SLIB] SPS30 SHDLC protocol:  2.0
-->[SLIB] SPS30 Library level :  1.4
Sending:  0x7E 0x00 0xD3 0x00 0x2C 0x7E
Received: 0x7E 0x00 0xD3 0x00 0x00 0x2C 0x7E length: 6

-->[SLIB] SPS30 Detected SPS30 via UART. 
Sending:  0x7E 0x00 0x00 0x02 0x01 0x03 0xF9 0x7E
Received: 0x7E 0x00 0x00 0x00 0x00 0xFF 0x7E length: 6

-->[SLIB] SPS30 Measurement OK 
-->[SLIB] UART detected SPS30 sensor :)
-->[SLIB] UART detected:  SENSIRION
[... 249 lines cut ...]
Sending:  0x7E 0x00 0x00 0x02 0x01 0x03 0xF9 0x7E
Received: 0x7E 0x00 0x00 0x43 0x00 0xBC 0x7E length: 6

43 : state error
instruction failed
-->[SLIB] UART sensors count:  0
-->[SLIB] AM2320 read > done! 
-->[SLIB] PM1:003 PM25:006 PM10:008 CO2:0000 CO2humi:0.000000% CO2temp:0.000000°C H:39.000000% T:26.600000°C
Sending:  0x7E 0x00 0x00 0x02 0x01 0x03 0xF9 0x7E
Received: 0x7E 0x00 0x00 0x43 0x00 0xBC 0x7E length: 6

43 : state error
instruction failed
-->[SLIB] UART sensors count:  0
-->[SLIB] AM2320 read > done! 
-->[SLIB] PM1:003 PM25:006 PM10:008 CO2:0000 CO2humi:0.000000% CO2temp:0.000000°C H:39.000000% T:26.600000°C
-->[IFDB] write done.
Sending:  0x7E 0x00 0x00 0x02 0x01 0x03 0xF9 0x7E
Received: 0x7E 0x00 0x00 0x43 0x00 0xBC 0x7E length: 6

43 : state error
instruction failed
-->[SLIB] UART sensors count:  0
-->[SLIB] AM2320 read > done! 
-->[SLIB] PM1:003 PM25:006 PM10:008 CO2:0000 CO2humi:0.000000% CO2temp:0.000000°C H:39.000000% T:26.600000°C

hpsaturn, 2021-09-24:
Thanks @arttesano for this complete report. I edited the previous message only for put the ouput into a block. I'm goint to replicate in my environment and also I will release a version in the OTA dev channel. Please subscribing this device to OTA dev channel with the next zip. With that we will test more quickly.

https://t.me/canairio/15342

closed 2021-09-24 by hpsaturn