kike-canaries/canairio_firmware #130: Set Wifi credentials without Android?

frank-f, 2021-10-04:
Yesterday I wanted to give CanAirIo a try, but got stuck at the point where I wanted it to connect to my Wifi. I have no Android phone and could not find a way to enter my Wifi credentials any other way?!
Have I been just too blind to see it or is there really no other way? Is the rest of CanAirIo maybe also useless without the Android App and I can scrap further attempts altogether?

hpsaturn, 2021-10-04:
Thanks Frank, is true, for now only with Android CanAirIO works. But we already working on iOS version. If you maybe can help on this development, we will be very happy. 

Is the next repository:
https://github.com/kike-canaries/canairio_android/tree/devel

The last pull request and goals:
https://github.com/kike-canaries/canairio_android/pull/110

Discussion:
https://github.com/kike-canaries/canairio_android/issues/90

The current goal in CanAirIO project is the mobile mode, because for fixed stations a think maybe exists a better alternatives than CanAirIO. If you know iOS programing or knows anyone that know it, please be free to invited it to CanAirIO development.

frank-f, 2021-10-04:
> Thanks Frank, is true, for now only with Android CanAirIO works. But we already working on iOS version. If you maybe can help on this development, we will be very happy.

Unfortunately I am not an iOS or Java developer. I wouldn't even know where to start, so I am probably no help in that aspect. I'd be happy to help testing, once the App is far enough to be installable though.

> The current goal in CanAirIO project is improve the mobile mode, because for fixed stations a think maybe exists a better alternatives than CanAirIO. If you know iOS programing or knows anyone that know it, please be free to invited it to CanAirIO development.

I started looking at CanAirIO since it seems to implement lots of interesting sensors in one project and it even has some basic display support. Which other projects to you have in mind, that I could try?

hpsaturn, 2021-10-05:
Thanks for your positive feedback. For example a alternative to try to give more support to CanAirIO without mobile phone dependency is the next project:

https://github.com/hpsaturn/CanAirIO_fixed_station

Maybe you can try to improve that. I don't know, which is your skills?

hpsaturn, 2021-12-10:
> Yesterday I wanted to give CanAirIo a try, but got stuck at the point where I wanted it to connect to my Wifi. I have no Android phone and could not find a way to enter my Wifi credentials any other [...]

Hi @frank-f, we have some good news, One developer is working on a iOS version with web technologies, maybe you can review this thread in our Telegram channel (en/es):

https://t.me/canairio/19361

hpsaturn, 2023-06-05:
With the new CanAirIO CLI is possible set the WiFi credentials and other settings without Android app or without any mobile phone, you able to use it via Chrome browser. More details here:

https://canair.io/docs/cli.html

closed 2023-06-05 by hpsaturn