kike-canaries/canairio_firmware #177: Add sensor Cubic PM1006K

erguro1973, 2021-12-22:
Please add PM sensor manufacturer CUBIC Model PM1006k

- Built-in MCU, directly output particle mass concentration (μg/m³)
- Optimized lens design, the sensor can identify 0.3μm particle
- High accuracy, high stability, quick response
- Temperature compensation with the whole measurement range
- Auto lens dust deposition correction
- Compact structure, light weight and easy for installation

Link to datasheet:  [http://www.jdscompany.co.kr/download.asp?gubun=07&filename=PM1006_LED_PARTICLE_SENSOR_MODULE_SPECIFICATIONS.pdf](url)

hpsaturn, 2022-09-01:
Hi @erguro1973 good news, we finally did it, we released a new version that has support to this sensor:
https://canair.io/docs/canairio_ikea.html

Thanks!

closed 2022-09-01 by hpsaturn

erguro1973, 2022-09-01:
Great Job!! Congrats for the big efforts