kike-canaries/canairio_firmware #206: Crash witth sleep

roberbike, 2022-02-23:
When changue between ECO  and sleep , I have some resets of the boards, some times the board is bloqued, and other times there are not publications.
Both borards has the same sleep time, but pico d4 don´t has the correct publication time 180 seg.  Sample time is 15 seg.

board:  T-Display,  ESP PicoD4
firmware version: rev904
mobile app version: rev606  Phone ALE-L21
sensors: (sps30 by i2c)
OLED: yes
TFT: yes
Power: T display with Battery/pico with USB
```

## Additional context

![image](https://user-images.githubusercontent.com/20878412/155405850-a0b6f985-82c6-44e3-b99b-146a9321b9b5.png)
![image](https://user-images.githubusercontent.com/20878412/155406074-6827f482-ad35-42ea-9570-1430c04c8c70.png)

roberbike, 2022-02-23:
[image]
See the blue line. Sample time 15 seg, sleep time 180 seg. When I set ECO off, the publication time is not 180 seg. I can changue to 120, 60, etc and nothing change, but when I changue again to ECO on, the publication time is 180 seg. SOmething is wrong with oled version.

hpsaturn, 2022-02-24:
Please try to test again with the rev905

closed 2022-03-11 by hpsaturn