kike-canaries/canairio_firmware #216: Allow for configuration of offsets for measured values

hapklaar, 2022-05-07:
It would be good to be able to set offsets for measured values, for example temperature offsets to compensate for too high readings.

hpsaturn, 2022-05-07:
@hapklaar thanks for your contribution (Github Sponsors) and your feedback. Right now some sensors has offsets, from the Android app, in settings->Mobile station you can find these settings. On the other hand, I don't know how to do high readings offset. Maybe you can extend or explain it?

hapklaar, 2022-05-07:
For example the device is showing a temperature of 26.2 C, while it is actually 19.5 inside at the moment. Also CO2 outside levels are calibrated to 400ppm minimum value, while currently (2022) outside air has a minimum of around 419ppm, causing the device to always show a too low value.

If it is possible to calibrate to custom values, or apply offsets to the values the device is measuring (in firmware or firmware configuration), this could be corrected.

Another measuring device project I've been following does something similar, maybe you have seen it?
https://github.com/melkati/CO2-Gadget

PS I wasn't aware of the offset configuration in the app. I have tried it now, and at first I thought it didn't work as it didn't seem to have an immediate effect on what was shown on the LCD. Now I see it seems to adjust slowly, perfect!

hpsaturn, 2022-05-07:
> For example the device is showing a temperature of 26.2 C, while it is actually 19.5 inside at the moment. Also CO2 outside levels are calibrated to 400ppm minimum value, while currently (2022) outside air has a minimum of around 419ppm, causing the device to always show a too low value.

Well, the current firmware supports it, temperature offset and CO2 calibration to do it in outdoors. But it's true, some CO2 sensors only have support for 400ppm calibration. The Sensirion SCD30 supports custom calibration value, the old sensors not. I'm going to put this requirement to the roadmap. Thanks. 

> 
> If it is possible to calibrate to custom values, or apply offsets to the values the device is measuring (in firmware or firmware configuration), this could be corrected.

Yes, it is possible. Temperature and Altitude, via Bluetooth from the CanAirIO Android app. But yes, we need the CO2 offset feature.

> 
> Another measuring device project I've been following does something similar, maybe you have seen it? https://github.com/melkati/CO2-Gadget

Yes, I know this, I did some pull request or contributions to this project, and also I know the developer. CO2-Gadget use the CanAirIO Sensors Library, that I did with others CanAirIO developers

Here I referenced it, in the [CanAirIO Sensors Library repository](https://github.com/kike-canaries/canairio_sensorlib#projects-using-this-library)

> 
> PS I wasn't aware of the offset configuration in the app. I have tried it now, and at first I thought it didn't work as it didn't seem to have an immediate effect on what was shown on the LCD. Now I see it seems to adjust slowly, perfect!

Yes, some sensors has a proprietary or internal algorithm and the offset has a tiny lag. With external or normal sensors, the current CanAirIO Sensors Library apply it direct (addition or subtraction in hot mode)

Thanks again for your comments and feedback.

hapklaar, 2022-05-07:
Thank you. I'll close this issue as what I requested is already possible. Looking forward to new developments!

closed 2022-05-07 by hapklaar

hpsaturn, 2022-05-08:
But, for now is not possible de CO2 offset, maybe you can report a new issue with type "feature request" with that? Thanks

hapklaar, 2022-05-08:
Yes, will do