kike-canaries/canairio_firmware #229: Espressif v5.0.0 GPIO issues

hpsaturn, 2022-07-28:
## Overview

Testing the last devel branch with Espressif framework 5.0.0 we have a weird output with GPIO.

## Debug output

```bash
12:42:40.940 > E (87103) gpio: gpio_set_level(226): GPIO output gpio_num error
12:42:41.192 > E (87355) gpio: GPIO can only be used as input mode
```

## Device config

```
board: M5StickC Plus
firmware version: `devel`
mobile app version: revxxx
sensors: with and without sensors
Power: USB
```

hpsaturn, 2023-01-26:
Maybe in relation with issue on #178

Jason2866, 2023-01-27:
Why using the old outdated framework 5.0.0? By doing this you use a old Arduino core version. There was a bug in GPIO settings, fixed in actual core.

hpsaturn, 2023-01-27:
Thanks so much, it is true, it resolve the GPIO issue, and the execution messages. Now I need research why with this toolchain my firmware go to freeze. But sounds that without any doubts is better this version :D

closed 2023-01-27 by hpsaturn