kike-canaries/canairio_firmware #239: Issue with temperatures below Zero °C

Marvin-1802, 2022-11-05:
## Overview
When temperatures drop below  -0.9°C  the the big readout shows large 5 digit numbers and it stops broadcasting data.  
(I tested just the AHT10 sensor in the freezer, the ESP board and battery remained at room temp.)

Also: this board is giving false (0.0) PM2.5 sensor data to Grafana/Influx.  I do not have a PM2.5 sensor connected
 I am unable to connect via UART from IKEA PM2.5 sensor to this board, and cannot connect to the IKEA PM2.5 Cubic sensor directly.

## Debug output
Not available.

## Device config
UART: Auto
Sample time: 60 sec
Debug: Enabled

```
board:  TTGO T-Display
firmware version: TTGO_TDISPLAY rev918(prod)
mobile app version: v0.8.4 rev612  - MotoG(5)Plus
sensors: AHT10 (i2c)
OLED: yes
Power: Battery
```

Marvin-YYC, 2023-04-12:
> Please, can you test this issue with the last testing version of CanAirIO. Thanks!

This issue with temperatures below zero still persists with ver: 918 and testing ver: 943 using only AHT10 temp/humi sensor.

hpsaturn, 2025-03-02:
This issue seems that was resolved. I checked and the wrong cast that we had, now it was removed.

closed 2025-03-02 by hpsaturn