kike-canaries/canairio_firmware #244: Support for https://sensor.community/ (Luftdaten)

stsdc, 2022-12-18:
Wondering if sending data to third party services like https://sensor.community/ (Luftdaten)  is on the ToDo list?

roberbike, 2023-01-02:
This is the roadmap: https://github.com/orgs/kike-canaries/projects/1

hpsaturn, 2023-01-03:
Thanks @roberbike. This roadmap is a little outdated, but I want to add that the Sensor Community support ever it has always been in the roadmap. For the moment yes, we can change the config to write via MQTT, but the idea is have a more easy switch or setting to publish to Sensor Community @stsdc

hpsaturn, 2023-02-12:
Hi @stsdc 
I want integrate a new alternative to post to Luftdaten. I reviewed that they have an API REST with POST for publish the sensor data, but maybe do you know if they have a alternative using MQTT? 

I did this question regarding a new CanAirIO CLI that I'm writing now, and I will glad if maybe we integrate this, like a new  cloud alternative in this new serial setup CLI of CanAirIO:

[image]

stsdc, 2023-02-12:
I can presume, that they have only REST api. Since that's only thing they describe in the docs and everyone using only that.