kike-canaries/canairio_firmware #246: Would like to see the option to add IKEA vindriktning PM2.5 sensor to the TTGO T-display

Marvin-1802, 2022-12-20:
Currently it is not possible to add the IKEA vindriktning PM2.5 sensor via UART to CanAirIO using the TTGO T-display.  
I would really like to see this option added, as I have purchased five IKEA IAQ devices and five T-displays that I would like to connect to them. 
Thanks :)

roberbike, 2023-01-02:
Ver. 922 is working with this setup.
Use the gpio 12 and 13 for uart comunication.

Marvin-1802, 2023-01-04:
I can't get this to work. I am connecting the IKEA sensor directly to the TTGO T-display as the uart connection didn't work with my Heltec board, but worked with Heltec board when connected without IKEA board uart.
It is still not recognizing the IKEA PM2.5 sensor.  I was able to obtain a buck booster, but it is different than the one you recommended. Using pin 27 for the booster and the 5v output from the booster to the sensor doesn't work for me.

roberbike, 2023-01-04:
Show a picture of the booster you have connected and how you have connected it.

Marvin-1802, 2023-01-04:
@roberbike I figured out the problem. . . the buck booster I have can't handle the load.  5.0v output unloaded but when connected to PM sensor only 2.34v output.  I'll have to wait a month for buck boosters from China. :(
Thanks

Marvin-YYC, 2023-01-28:
**TTGO T-Display (Bike) r926**
I received my 5v buck boosters which appear to work fine @ 5.2v while connected to the IKEA PM2.5 sensor Unfortunately  Canairio still won't recognize it.  When I connect _AHT10_ to pins 21/22 temp+humi are measured, but when I connect PM2.5 sensor to pins 21/22 with the _AHT10_ ,temp+humi show correctly but no PM2.5.  If I connect PM2.5 to 21/22 sensor alone it is not recognized.  I have used this sensor with Heltec board successfully so I know the sensor is good.

roberbike, 2023-01-28:
- You have a big problem with the electrical connections. Ikea sensor can not be connectec to 21/22 pins. This sensor is nor an I2c sensor. Don´t do this, please. Read my previous messages carefully. Never will be recognized in this pins.
- You can not mix 2 sensors, with the same pins 21/22 and diferent voltage.
- I made you an schema:
- 
[image]

Marvin-YYC, 2023-01-29:
Oh Sorry @roberbike . .  I am getting old.  I forgot about your previous statement regarding pins 12/13.  Working on other things while I was waiting for buck boosters.  

Thank you for the schematic.  I see the problem now. . . 
I have three IKEA sensors and only one of them works with the IKEA uart board connection.  I connected my Heltec boards directly to the sensors without the IKEA board and they were working that way, but couldn't get AHT10 T+H sensors to work with the Heltec boards so wanted to get the T-display boards to work the same way with PM2.5 and AHT10.  I guess it is not possible to connect the IKEA sensors directly to the T-display without the IKEA uart board..  Pins 12/13 won't work for my IKEA sensors as the uart board doesn't work correctly.  :/

Marvin-YYC, 2023-01-29:
I connected another the IKEA sensor, with the only working ikea uart board that I have, to the T-display and it won't work either.  
Initially the buck booster was over loaded (3.4v) so I disconnected the circ fan in the ikea sensor, but the canairio board still won't detect the PM sensor.  The uart connection works fine with the Heltec board, I am powering the Heltec board from the uart connection through the ikea board.  Still can't get Canairio to recognize the AHT10 T+H sensor using pins 21/22.  ????
Just pulling my hair out now . . . ¯\_(ツ)_/¯

roberbike, 2023-01-29:
The Ikea sensor has 2 parts. One board with usb-C connector and the sensor Pm1006 with the fan.
The ikea circuit board, have 5 copper circules marked as ISPDA, REST, G, 5V.
You have to connect with wires solded, the 5V to 5V out of buck boost, G to G out of buck boost, and REST to 13 and ISPDA to 12 on T-display . The 3V and gnd on Tdisplay, to the imput of buck boost. Pin 27 to enable, of buck boost.
With the firmware  upper to 922 version, has to work.

Send me photo about your welded cables.
Connect only one usb cable to power the canairio. Try first the usb-c on ikea circuit board. With this connection there is not usb comunication to the pc, it is only for power the system.
When you has the PM sensor working, connect the AHT10 sensor. Go step by step, do not run.

roberbike, 2023-01-30:
If the step up output (5V) drops when you switch on the fan, this is not a good sign. You could clean the solder joints with IPA isopropyl alcohol, there is a lot of tin residue.
Check that, without the fan connected, the voltage at the step up output is 5V.
One last test before discarding the sensor, swap wires 12 and 13.

Marvin-YYC, 2023-01-31:
@roberbike 
Yes; some flux residue.
I tried another T-display board that is running  ver: **r918** and IKEA sensor with uart is now working correctly with fan, same buck booster and AHT10 is working too.  :)
After I swapped the board I noticed that the **r926** board had a low battery showing only 3.50v and no bars.  I checked the 5.0v o/p pin and it has only 3.3v?!?!  Possibly a faulty board or I damaged it.  :/
Thanks for your help Roberbike.

hpsaturn, 2023-02-11:
then, we can close this ticket? what do you think? On the other hand, remember that the Testing version now has the option to change the TX/RX pins via setup, via serial console.

[image]

Marvin-YYC, 2023-02-11:
My new TTGO T-Display board is now working well with AHT10 and IKEA PM2.5 Sensor.  
Thanks all for the assistance.

closed 2023-02-11 by Marvin-YYC