kike-canaries/canairio_firmware #247: wifi sniffer

haxzor420, 2023-01-23:
Hello, 

How does the wifi sniffer feature work/how do you interpret it?

Thanks.

Mike

hpsaturn, 2023-01-24:
Hi, thanks for ask, well it is a basic WiFi sniffer derivative of project [PaxCounter](https://github.com/cyberman54/ESP32-Paxcounter). It use the both Bluetooth and WiFi radios to try to count people around you. Some implementation using also a small data base of routers IDs or other devices IDs to try to remove these kind of devices of the general counter. In CanAirIO the Pax unit is used to try to measure the impact over people, of bad air quality conditions into some spaces, like public transportation for example. The PAX is used like other sensor unit, and you can enable/disable it. Only works well in mobile station mode (Bluetooth+mobile phone), because on fixed station using WiFi, it will disable because is unstable with the default WiFi connection.

hpsaturn, 2023-01-24:
I did some posts in twitter some moths ago:

In train:
https://twitter.com/hpsaturn/status/1412442750545498116

In airplane:
https://twitter.com/hpsaturn/status/1524022894082281472

closed 2023-03-02 by haxzor420