kike-canaries/canairio_firmware #269: Disconnecting and removing wifi AP does not work via CLI ( Chrome)

Geelhem, 2023-10-17:
## Overview
I have the board connecting to wifi but I want to remove AP and add a new one the commands are accepted but list of neworks is not updated. 

I have scan and attenmpted to leave MG1.0 to MG2.0 password is saved but board does not attemot to connect to it

## Debug output

= CanAirIO Setup ==

-->[POWR] CPU Speed: 240 MHz
-->[OGUI] display config ready.

SerialTerm v1.1.2
(C) 2022, MikO - Hpsaturn
  available commands:

-->[BATT] ADC eFuse Vref:1093 mV
-->[INFO] ESP32MAC		: B0:B2:1C:0A:FC:56
-->[INFO] Hostname		: CanAirIOC56
-->[INFO] Revision		: r951
-->[INFO] Firmware		: 0.5.8
-->[INFO] Flavor  		: ESP32DEVKIT
-->[INFO] Target  		: prod
-->[INFO] == Detecting Sensors ==
-->[INFO] Sensorslib version	: 0.6.7
-->[INFO] enable sensor GPIO	: 27
-->[INFO] config UART sensor	: SPS30
-->[SLIB] temperature offset	: 0.00
-->[SLIB] altitude offset   	: 0.00
-->[SLIB] sea level pressure	: 1013.25 hPa
-->[SLIB] only i2c sensors  	: true
-->[SLIB] sensor registered	: SPS30  	:D
-->[SLIB] sensor registered	: AM232X  	:D
-->[SLIB] Sensors i2c count 	: 2 (SPS30,AM232X,)
-->[INFO] Sensors detected	: 2
-->[INFO] registering units	:
-->[INFO] restored saved unit	: PM2.5
-->[SLIB] Sensors units count	: 6 (PM1,PM2.5,PM4,PM10,T,H,)
-->[WDOG] watchdog check each	: 120 seconds.
-->[WIFI] connecting to wifi	: MG1.0...............................fail!
[E][WIFI] disconnected!
-->[INFO] InfluxDb cloud 	: enabled
-->[INFO] WiFi current config	: enabled
-->[BTLE] Bluetooth GATT 	: ready for config client!
-->[INFO] sensors units count	: 6
-->[INFO] show unit selected 	: PM2.5
-->[HEAP] sizeof sensors	: 0996b
-->[HEAP] sizeof config 	: 0296b
-->[HEAP] sizeof GUI    	: 0292b

==>[INFO] Setup End. CLI enable. Press ENTER  ===

	help		show detail usage information
	setSSID		set the Wifi SSID
	setPASW		set the WiFi password
	connect		save and connect to WiFi network
	list		list saved WiFi networks
	select		select the default AP (default: last)
	mode		set the default operation single/multi AP (slow)
	scan		scan WiFi networks
	status		WiFi status information
	disconnect	WiFi disconnect
	delete		remove saved WiFi network by SSID

	reboot		perform a ESP32 reboot
	clear		factory settings reset. (needs confirmation)
	debug		enable debug mode
	stime		set the sample time (seconds)
	spins		set the UART pins TX RX
	stype		set the sensor type (UART)
[... 272 lines cut ...]
st> .................fail!
[E][WIFI] disconnected!
-->[WIFI] connecting to wifi	: MG1.0.. done.
-->[WIFI] device network IP	: 192.168.2.106
-->[WIFI] publish interval 	: 10 sec.
-->[FOTA] remote OTA update 	: not need update
[E][MQTT] Anaire publish error 	: -3
-->[MQTT] 192.168.2.23	: connected!
-->[MQTT] incoming msg: homeassistant/status - offline
[E][MQTT] Anaire publish error 	: -3
-->[MQTT] 192.168.2.23	: connected!
Terminal disconnected: NetworkError: The device has been lost.
>
## Device config

**If apply**, please write the next information, you can have it on the settings section `Device info` on the CanAirIO app.

```
board: please choose any  ESP32DevKit, and similar behaviour on M5stick+
firmware version : Latest stable
mobile app version: revxxx
sensors: I2C SPS30
OLED: yes
Power: USB
```

hpsaturn, 2023-10-18:
tip: to work with the CLI without debug messages, first disable the debug mode with `debug 0`

hpsaturn, 2023-10-18:
Maibe it is a duplicated ticket:
https://github.com/kike-canaries/canairio_firmware/issues/250

Please let me know if is the same behavior. I'm trying to understand what happen with this old issue.

hpsaturn, 2025-03-02:
Fixed on dev channel. Please reopened it if not. Thanks

closed 2025-03-02 by hpsaturn