kike-canaries/canairio_firmware #286: Seeed Xiao ESP32c3 randomly disconnects

Marvin-YYC, 2024-12-30:
## Overview
I now have two Seeed Xiao CanAirIO devices running both using the same Wi-Fi, and they both randomly stop reporting to the influxdb.
I have a bike device also running, and I do not see this problem.  They do not do it at the same time.

## Debug output
I do not have them connected to a pc

## Device config

**If apply**, please write the next information, you can have it on the settings section `Device info` on the CanAirIO app.

```
board: please choose any TTGO T7, TTGO T-Display, ESP32DevKit, WemosOLED, etc
firmware version: ESP32C3 SEEDX
mobile app version: n/a
sensors: IKEA Vindstyrka and BME680
OLED: No
Power: USB
```

hpsaturn, 2024-12-31:
Thanks for report it. These XIAO what version are using?

Marvin-YYC, 2025-01-04:
Sorry I forgot to send :/

R- 982
firmware- 0.7.0

hpsaturn, 2025-01-04:
I don't have this specific board, the XIAO C3, but I notice two things:

- The current devel branch has a old version of Espressif platform. I updated it to 6.9.0
- The other thing is that I have a strange workaround that I did for the first migration to the ESP32C3:

```cpp
#ifdef FAMILY
  if (FAMILY == "ESP32-C3") WiFi.setTxPower(WIFI_POWER_8_5dBm);  // TODO: uggly workaround for some C3 devices
#endif
```

This weird workaround is in the CLI layer. In the "ESP32 WIFI CLI" library.
Let me change this, and I'll to notify to you for please do a new test.

Marvin-YYC, 2025-02-01:
I am still having trouble with this IKEA Xiao C3 CanAirIO.  It still occasionally seems to disconnect from Wi-Fi or otherwise fail.  It is not connected to my PC when this happens, so I can't provide any log when it occurs.  

Yesterday it stopped updating, but still appeared to be connected.  All the data was straight line.  Stuck.  When I power cycled the device all data dropped to zero for about 30 min.  I power cycled it again and it returned to normal.  The other Xiao C3 that I have outside with a BME680 hasn't had any problems since the latest update.  

I checked that it wasn't my Wi-Fi or other issues.  No idea what is going on, unless it is the IKEA board causing the issue.  I have the Xiao powered through USB directly, and the IKEA device is powered by the Xiao.  

[image]

[image]

hpsaturn, 2025-02-01:
Thanks @Marvin-YYC for this complete report. I already fixed the typo issue, thanks again.

Well, I don't know too what happen with all here. I haven't a XIAO C3, but I noticed that one of my devices, with the last dev version, had a problem similar. In this case it was a T-Display board, and only with a manual reset it returned again to publish. But right now it has weeks working well.. I don't know.. Well, please let me know any update. I will looking to try to catch what happen..

Marvin-YYC, 2025-02-17:
Since the update the Xiao device that I have outside has been stable, but the Xiao device that is connected to he IKEA IAQ sensor still seems to disconnect or go straight line occasionally.  
I think that there may be interference from the radio chip inside the IKEA device or possibly from other IOT devices in my home.  I have moved other IOT devices away from the IKEA Xiao device and the events seem to occur less often.

closed 2025-02-17 by Marvin-YYC