kike-canaries/canairio_firmware #297: Both my Xiao C3 devices went offline today

Marvin-YYC, 2025-03-03:
## Overview
Both my Xiao C3 devices went offline today March 2/25 at  13:00 MST / March 3/25 08:00 UTC
I connected them to my PC and they were not behaving normally and were not detecting any sensors.  One is connected to an IKEA Vindstyrka and the other is connected to a BME680.  

## Debug output

[esp-web-tools-logs (1).txt](https://github.com/user-attachments/files/19047053/esp-web-tools-logs.1.txt)

## Device config

**If apply**, please write the next information, you can have it on the settings section `Device info` on the CanAirIO app.

```
board: Seeed Xiao ESP32 C3
firmware version: r987
mobile app version: 
sensors:  IKEA Vindstyrka /  BME680
OLED: no
Power: USB
```

hpsaturn, 2025-03-05:
With the last OTA this issue was resolved?

Marvin-YYC, 2025-03-06:
Both of my Xiao devices are indicating 987.  I updated the IKEA device three times from the website, but it is also still on 987.

Should it update by OTA?

hpsaturn, 2025-03-06:
Yes, is weird, I received the OTA some days ago. Mmm, just now I updated the webinstaller with the last version. Please, if could try it. Thanks!

Marvin-YYC, 2025-03-08:
I was able to update to r988 using the web installer.  Sensors are now showing up.

Everything looks good, devices have appeared on Grafana again.  

Many thanks.

closed 2025-03-08 by Marvin-YYC